thanhndv212/figaroh-plus #1: Is this an upgraded version of Figaroh?

Nnboy, 2025-04-20:
Hello Development Team,
I found your code on [gitlab](https://gitlab.laas.fr/gepetto/figaroh) before, is this repository an upgraded version?

thanhndv212, 2025-04-20:
Hello. 
Thanks for your interest in the project. In short, yes this aims to be an upgraded, truly easy-to-use version. 
I stopped participating in maintaining the one on gitlab, so I forked and also am trying to improve it. Currently

thanhndv212, 2025-04-20:
Hello. 
Thank you for your interest in this project. 
In short, yes I forked the repo and aim to make it an upgraded version of the one on gitlab, since I stopped participating maintenance of that on git lab (they are separately developed from now on). 

Currently, I have made some improvement:
-  replaced the cmake install with an simplified installation with conda 
- added a documentation page along with the already published paper

In the future, I have plan to:
- To add a step-by-step guide on each type of robot
- Identification of actuators
- Identification/Calibration of closed-loop robots
- Integration to RL framework

So, all contributions and feedbacks are welcome.

thanhndv212, 2025-09-05:
I have upgraded and officially release it. Now you can:
- Install from pypi
- Better API
- Separate examples dir to an individual repo
If you can, please give it a try and let me know any problem. Thanks.

Nnboy, 2025-09-05:
@thanhndv212 

Thank you for the detailed explanation and for sharing the significant improvements you've made! It's great to hear about the progress, especially the simplified installation via PyPI and the API enhancements.

I'll definitely give the new PyPI version a try in the near future and put it through its paces. Once I've had a chance to test it out with some use cases, I'll be sure to provide you with feedback here.

Looking forward to exploring the upgraded Figaroh-Plus!

closed 2025-12-07 by thanhndv212